Guilouz/Klipper-Flsun-Speeder-Pad #2: Root via update.sh still work?

arraylabs, 2022-09-15:
I'm not having any luck with the update.sh appearing to do anything. Does this method still work with recently delivered printers/speederpads?

Guilouz, 2022-09-15:
Just fixed this. Re-download zip file.

closed 2022-09-15 by Guilouz

arraylabs, 2022-09-15:
Thanks, unfortunately it still doesn't appear to do anything. I'm unable to log into the pad with the provided username and password. I also don't see the .over file on the USB after booting the pad with the stick inserted which kinda implies the update.sh script isn't getting run.

Guilouz, 2022-09-15:
Try to re-download new zip, it have been updated to set password for pi user to flsun.

arraylabs, 2022-09-16:
Sorry to say but still no luck. I appreciate your time trying to sort it out.

arraylabs, 2022-09-16:
Just to confirm where the directions say "root folder" that should be the same level as the cura 4 & 5 installation files, correct?

Guilouz, 2022-09-16:
Not understand what you mean with Cura.

Root Acess is needed to change time zone, wifi location, klipper etc in SSH, nothing to see with Cura.

jFensch, 2022-09-16:
I am also experiencing the issue:
[image]

Guilouz, 2022-09-16:
It seems Flsun have changed something to block root access.

Brubrui, 2022-10-03:
At night, does the password seem to have changed because flsun is not giving access to the spider pad? Do you know anything about it?. thank you