Guilouz/Klipper-Flsun-Speeder-Pad #4: Erreur mise à jour Klipperscreen

ManuMod, 2022-10-20:
Après la modif de moonraker.conf (en PJ), le Hard recovery retourne l'erreur "Repo has not been verified, clone aborted".
[moonraker.conf.txt](https://github.com/Guilouz/Klipper-Flsun-Speeder-Pad/files/9834097/moonraker.conf.txt)

ManuMod, 2022-10-20:
J'ai suivi la procédure avec 3 instances.

Guilouz, 2022-10-20:
Suis la procédure avec une seule instance, il y a des problèmes de compatibilités avec les nouveaux chemins de moonraker changés récemment.

ManuMod, 2022-10-20:
OK, merci pour l'info.
Je voulais conserver la fonctionnalité pour plus tard, mais je peux effectivement débuter Klipper avec 1 instance et refaire une install si je trouve plus stable en multi instances. Tu envisages également de l'utiliser en multi instances ?

Guilouz, 2022-10-20:
Non c’est juste un pad de test, j’en ai un autre avec ma V400

closed 2022-10-20 by Guilouz

ManuMod, 2022-10-22:
Pour l'info, j'ai eu la même erreur avec 1 seule instance de Klipper, mais qui a été résolu après une mise à jour de Moonraker.

Guilouz, 2022-10-22:
Oui il y a eu des updates aujourd'hui pour Moonraker et notamment un correctif pour le multi instance.

ManuMod, 2022-10-22:
Effectivement, j'ai vu ton issue sur le sujet.
Je ne sais pas s'il y a un rapport avec les màj de Moonraker, mais j'ai également eu 1 erreur avec le service WebcamHD qui restait en état Failed. Ce que j'ai résolu à coup de plugin mainsail/crowsnest.

Guilouz, 2022-10-22:
J’étais pas sûr que crownest fonctionnait correctement sous Ubuntu. Pas de problème avec? La configuration de la cam avec le fichier de conf fonctionne ?

ManuMod, 2022-10-22:
Cela fonctionne tip top avec ma C310. J'ai repris la config par défaut en modifiant seulement la résolution pour 720HD. Aucune erreur à l'install par contre l'ajout auto de l'update n'écrit pas les lignes dans monraker.conf. Faut penser à les ajouter.