Guilouz/Klipper-Flsun-Speeder-Pad #5: Utiliser gcode sur clef USB

ManuMod, 2022-10-22:
Même si l'envoi direct du gcode sur Mainsail est l'utilisation principale, je souhaiterai pouvoir conserver la fonctionnalité de lecture des gcodes sur clef USB.

Le montage de la clef USB n'est pas automatique et une fois montée en SSH, elle n'est pas reconnue dans Mainsaill.

Faut-il mettre en place qq chose du style autofs ou script, associé à un symlink sur 1 dossier dédié dans pi/gcodes_files ?

Guilouz, 2022-10-22:
C’est pas le but de Klipper de print via clé USB ou microSD. C’est bien pour cette raison que ce n’est pas une fonctionnalité de Mainsail.
Il me semble que l’USB est monté uniquement au démarrage si une clé usb est insérée.
Je t’avouerais que je n’ai pas testé, j’envoie tous mes gcode depuis Cura directement sur le pad et ensuite je lance l’impression depuis le pad.

ManuMod, 2022-10-22:
Dsl, effectivement la clef est correctement montée au démarrage et même à chaud, mais dans /pi/gcode_files/USB-Disk alors que Mainsaill utilise /printer_data/gcodes.

Guilouz, 2022-10-22:
Merci de l'info, je vais rajouté ça au tuto.

closed 2022-10-22 by Guilouz

ManuMod, 2022-10-22:
Je pense qu'il serait plus propre de faire un symlink sur les dossiers gcode uniquement et cela permettrai pour le mode 3 instances (que je compte toujours essayer) de faire un symlink par printer_data sur le même dossier gcodes. Plus simple si une seule et même base de gcode, non?

Guilouz, 2022-10-22:
Oui ça peut être réalisable avec:

ln -s /home/pi/gcode_files/USB-Disk /home/pi/printer_data_1/gcodes/USB-Disk
ln -s /home/pi/gcode_files/USB-Disk /home/pi/printer_data_2/gcodes/USB-Disk
ln -s /home/pi/gcode_files/USB-Disk /home/pi/printer_data_3/gcodes/USB-Disk

J'ai pas encore vraiment poussé les recherches avec plusieurs instances.

ManuMod, 2022-10-22:
Effectivement ce serait LA méthode pour 3 instances.

Ce que j'évoquais était plutôt de limiter le symlink au /home/pi/gcode_files/, mais après mûre réflexion (ou plutôt 1 peu de repos ;-), c'est une bêtise pour les 3 instances, au vue des config dédiées aux imprimantes et/ou PLA.

Guilouz, 2022-10-22:
Oui car chaque imprimante est différente et n’utilisent pas forcément toutes les mêmes gcodes.
Il est préférable d’avoir pour chacunes d’elles leurs dossiers gcode.