Guilouz/Klipper-Flsun-Speeder-Pad #6: Timelapse pour 3 instances de Klipper

ManuMod, 2022-10-27:
Salut,

L'installateur de Timelapse ne gère pas le multi-instance de Klipper. Du coup, cela se passe manuellement sous SSH (source [https://github.com/mainsail-crew/moonraker-timelapse/issues/13](url)).

Exemple pour 3 instances de Klipper :

```
cd ~/
git clone https://github.com/mainsail-crew/moonraker-timelapse.git
ln -sf "/home/pi/moonraker-timelapse/component/timelapse.py" "/home/pi/moonraker/moonraker/components/timelapse.py"
ln -sf "/home/pi/moonraker-timelapse/klipper_macro/timelapse.cfg" "/home/pi/printer_1_data/config/timelapse.cfg"
ln -sf "/home/pi/moonraker-timelapse/klipper_macro/timelapse.cfg" "/home/pi/printer_2_data/config/timelapse.cfg"
ln -sf "/home/pi/moonraker-timelapse/klipper_macro/timelapse.cfg" "/home/pi/printer_3_data/config/timelapse.cfg"
sudo reboot
```

:warning: Ne pas oublier d'ajouter [timelapse] dans le fichier Moonraker.conf, ainsi que les lignes pour l'update.

Si le service WebcamHD ne fonctionne pas (comme moi :smile:) , possible d’utiliser le plugin de Mainsall : [https://github.com/mainsail-crew/crowsnest](url)

Guilouz, 2022-10-27:
Bien vu, j'utilise pas les timelapse alors je me suis pas penché dessus.

Guilouz, 2022-10-27:
J'ai déjà reporté l'issue ici : https://github.com/mainsail-crew/moonraker-timelapse/issues/71

closed 2022-10-27 by Guilouz