Guilouz/Klipper-Flsun-Speeder-Pad #8: ADXL345 issue ...

themacboy, 2022-10-29:
Excuse me, I doubt this is a problem with your config. But I will try to ask for help.

After running the TEST_RESONANCES AXIS=X, I get always this error:

[image]

Any idea? Is this speeder pad to weak?

themacboy, 2022-10-29:
[image]

Ummm xorg? why is this process running and having too much CPU load? lol

themacboy, 2022-10-29:
I just used:

sudo service KlipperScreen stop

to stop Xorg, but the problem still happens!

themacboy, 2022-10-29:
Solved !

I have reduced microsteps from 128 to 32. 

Too much data load to run a resonance test !

Guilouz, 2022-10-29:
Working fine for me with 128 microsteps. I have doing many tests for that without issue.

closed 2022-10-29 by Guilouz

themacboy, 2022-10-30:
Can you share the actual specs of your Speeder Pad? I like to compare with mine to check for different hardware versions.

Thanks in advance!

Guilouz, 2022-10-30:
description: Computer
    product: sun50iw10
    width: 32 bits
    capabilities: smp
  *-core
       description: Motherboard
       physical id: 0
     *-cpu:0
          description: CPU
          product: cpu
          physical id: 0
          bus info: cpu@0
          size: 1512MHz
          capacity: 1512MHz
          capabilities: fp asimd aes pmull sha1 sha2 crc32 cpufreq
     *-cpu:1
          description: CPU
          product: cpu
          physical id: 1
          bus info: cpu@1
          size: 1512MHz
          capacity: 1512MHz
          capabilities: fp asimd aes pmull sha1 sha2 crc32 cpufreq
     *-cpu:2
          description: CPU
          product: cpu
          physical id: 2
          bus info: cpu@2
          size: 1512MHz
          capacity: 1512MHz
          capabilities: fp asimd aes pmull sha1 sha2 crc32 cpufreq
     *-cpu:3
          description: CPU
          product: cpu
          physical id: 3
          bus info: cpu@3
          size: 1512MHz
          capacity: 1512MHz
          capabilities: fp asimd aes pmull sha1 sha2 crc32 cpufreq
     *-cpu:4 DISABLED
          description: CPU
          product: idle-states
          physical id: 4
          bus info: cpu@4
     *-memory
          description: System memory
          physical id: 5
          size: 976MiB
  *-usbhost:0
       product: SW USB2.0 'Enhanced' Host Controller (EHCI) Driver
       vendor: Linux 4.9.191 ehci_hcd
       physical id: 1
       bus info: usb@1
       logical name: usb1
       version: 4.09
       capabilities: usb-2.00
       configuration: driver=hub slots=1 speed=480Mbit/s
     *-usb
          description: USB hub
          product: USB2.0 Hub
          vendor: Genesys Logic, Inc.
          physical id: 1
          bus info: usb@1:1
          version: 85.38
          capabilities: usb-2.00
          configuration: driver=hub maxpower=100mA slots=4 speed=480Mbit/s
        *-usb
             description: USB hub
             product: USB2.0 Hub
             vendor: Genesys Logic, Inc.
[... 180 lines cut ...]
       logical name: /dev/input/event4
       capabilities: i2c
  *-input:4
       product: sunxi-ir-tx
       physical id: c
       logical name: ir_tx
       logical name: /dev/input/event2
       capabilities: platform
  *-network:0
       description: Ethernet interface
       physical id: d
       logical name: wlan0
       serial: 70:f7:54:cd:55:e1
       capabilities: ethernet physical
       configuration: broadcast=yes driver=wl driverversion=0 ip=192.168.0.36 multicast=yes
  *-network:1
       description: Ethernet interface
       physical id: e
       bus info: usb@1:1.1.2
       logical name: eth0
       serial: a2:0b:88:68:58:75
       size: 10Mbit/s
       capacity: 100Mbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8152 driverversion=v1.08.9 duplex=half link=no multicast=yes port=MII speed=10Mbit/s

themacboy, 2022-10-31:
Looks similar :( hahaha

Thx

fzechert, 2022-11-08:
> Solved !
> 
> I have reduced microsteps from 128 to 32. (one mate on chat warm me about that value)
> 
> Too much data load to run a resonance test !

Thanks, this worked for me as well, I used 64 microsteps in the configuration file.
After the calibration is done, I can set microsteps back to 128 right? Only a problem during the calibration.

Guilouz, 2022-11-08:
I have no issue with adxl and 128 microsteps.
No you must use settings you have used while Resonance testing

themacboy, 2022-11-08:
> I have no issue with adxl and 128 microsteps. No you must use settings you have used while Resonance testing

listen people! ;) lot of them having this issue while doing resonance check or while printing too.

Guilouz, 2022-11-08:
> > I have no issue with adxl and 128 microsteps. No you must use settings you have used while Resonance testing
> 
> listen people! ;) many of them have this problem while doing resonance check or also printing.
> 
> And sincerelly, I don't see any reason to use high microsteps.

I have 2 pad, one V400 pad on Ubuntu 20.04 LTS and one standalone pad on Ubuntu 22.04 LTS, I tested them both with resonance tests up to 130 Hz without problem (with rpi pico and pis module) and printed an object older than 17h.

I don't understand what's affecting this in both of you but it works well in other people as well.
However nothing prevents you from using 16 microsteps if you want ;) More steps potentially means a more precise machine.

themacboy, 2022-11-08:
> I don't understand what's affecting this in both of you but it works well in other people as well. However nothing prevents you from using 16 microsteps if you want ;) More steps potentially means a more precise machine.

I don't understand it either.

I personally lowered it to 32, and now it works fine.
Really, just a 3D printer we don't need this high precision and loss of torque

fzechert, 2022-11-09:
Thanks for your help and hints. Here is my solution that I managed to get working (kind of). If somebody stumbles upon this issue in the future, it might help.

Unfortunately, I did not have another Microcontroller that is compatible with Klipper other than the PI Zero. So my solution now was to set up Klipper completely on a Raspberry PI 4B 4GB, and not using the Speeder Pad from FLSUN.

This makes it possible to connect the ADXL to the GPIO PINs of the Raspberry PI directly, without the need of an intermediate Pi Zero. Configuration for this can be found in the Klipper Manual: https://www.klipper3d.org/Measuring_Resonances.html and https://www.klipper3d.org/RPi_microcontroller.html

Even with the Raspberry PI 4B 4GB I was not able to complete the SAHPER_CALIBRATE command successfully. It worked for the X Axis but the Y Axis failed at 124 Hz (with 128 micro steps). But since the test normally ends at 133 Hz, which is not so far from 124 Hz, I figured this is good enough.

So I ran the following commands in Klipper:

```
SHAPER_CALIBRATE AXIS=x
SHAPER_CALIBRATE AXIS=y FREQ_END=123
``` 

This worked fine and I got the following results:

```
[input_shaper]
shaper_type_x = mzv
shaper_freq_x = 53.4
shaper_type_y = mzv
shaper_freq_y = 56.2
```

Then I disconnected the Raspberry PI and connected again with the Speeder Pad. Just Copy those configurations to the `printer.cfg` in the SpeedPad, and done.

I still don't know why we have so many problems with the input shaper. But this is good enough for me now. 

If anybody has an Idea on how to trouble shoot this further and find / fix the underlying problem, let me know. I will be glad to help in troubleshooting. But until then, I am out of ideas as well.

:smile: