Guilouz/Klipper-Flsun-Speeder-Pad #11: Update docs

piwi3910, 2022-11-02:
Please update the docs, to reflect the V400 motherboard needs a max 8GB sdcard, with bigger ones, you will not be able to format it correctly.

themacboy, 2022-11-02:
Are you sure? I only own 16gb micro sd cards, and remember that I have updated my MKS Robin nano V.2 a few times. 

But not totally sure now.

piwi3910, 2022-11-02:
I'm sure a 32GB doesn't work, tried multiple
Tried it with a 4 and 8 an both work. 16 I didn't try as I didn't have any

[EXTERNAL EMAIL]

Internal Use - Confidential

Guilouz, 2022-11-03:
I can flash my V400 with my 16Gb or 32Gb microSD without issue.

closed 2022-11-03 by Guilouz

fzechert, 2022-11-06:
I can confirm this issue with my V400 printer. Flashing the SpeederPad worked with 32GB / 64GB micro SD card. Flashing the Robin_nano35.bin firmware to the board installed in the top of the printer did not work with 32GB / 64GB micro SD card. I had an old 64MB SD-Card with a SD to micro SD adapter that worked fine.

Maybe just add a hint, if the flashing does not work, use a <= 16GB SD Card.

Anyway, thanks for this great manual and your effort @Guilouz