Guilouz/Klipper-Flsun-Speeder-Pad #12: crowsnest install docs incomplete

piwi3910, 2022-11-03:
There are some steps missing for crowsnest

1) the default crowsnest service looks for the config file at /home/pi/klipper_config/crowsnest.conf
This means that the template that you copy over in your step:

"cp /home/pi/crowsnest/sample_configs/mainsail_default.conf /home/pi/printer_data/config/crowsnest.conf"
is not loaded

as you can see from this log:

[11/03/22 08:43:33] crowsnest: Version Control: ffmpeg is up to date. (4.2.7-0ubuntu0.1)
[11/03/22 08:43:33] crowsnest: INFO: Print Configfile: '/home/pi/klipper_config/crowsnest.conf'
[11/03/22 08:43:33] crowsnest:          [crowsnest]
[11/03/22 08:43:33] crowsnest:          log_path: ~/klipper_logs/crowsnest.log
[11/03/22 08:43:33] crowsnest:          log_level: verbose
[11/03/22 08:43:33] crowsnest:          delete_log: false
[11/03/22 08:43:33] crowsnest:
[11/03/22 08:43:33] crowsnest:          [cam 1]
[11/03/22 08:43:33] crowsnest:          mode: mjpg
[11/03/22 08:43:33] crowsnest:          port: 8080
[11/03/22 08:43:34] crowsnest:          device: /dev/video0
[11/03/22 08:43:34] crowsnest:          resolution: 640x480
[11/03/22 08:43:34] crowsnest:          max_fps: 15

the fix is:

sudo nano /etc/systemd/system/crowsnest.service
 change the line Environment to reflect the right path
Environment=CROWSNEST_CONFIG=/home/pi/printer_data/config/crowsnest.conf
systemctl daemon-reload
and restart your crowsnest service from mainsail or systemd
rm -rf /home/pi/klipper_config

then logs show correct:

[11/03/22 08:47:44] crowsnest: Version Control: ffmpeg is up to date. (4.2.7-0ubuntu0.1)
[11/03/22 08:47:44] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[11/03/22 08:47:44] crowsnest:          [crowsnest]
[11/03/22 08:47:44] crowsnest:          log_path: ~/klipper_logs/crowsnest.log
[11/03/22 08:47:44] crowsnest:          log_level: verbose
[11/03/22 08:47:44] crowsnest:          delete_log: false
[11/03/22 08:47:44] crowsnest:
[11/03/22 08:47:44] crowsnest:          [cam 1]
[11/03/22 08:47:44] crowsnest:          mode: mjpg
[11/03/22 08:47:44] crowsnest:          port: 8080
[11/03/22 08:47:44] crowsnest:          device: /dev/video0
[11/03/22 08:47:44] crowsnest:          resolution: 1920x1080
[11/03/22 08:47:44] crowsnest:          max_fps: 25
[11/03/22 08:47:44] crowsnest:          v4l2ctl: focus_auto=0,focus_absolute=50,zoom_absolute=100,exposure_auto_priority=0,exposure_auto=1,exposure_absolute=500,brightness=150,contrast=128,saturation=128,sharpness=128,white_balance_temperature_auto=0,white_balance_temperature=3250

now if you run 3 instances: copying over the template file 3 times doesn't seem to work well.
as crowsnest only runs once, and takes all camera's it can find.
So it would be better to copy it once, configure all your camera's in the config of instance 1
and then in the other instances select the right cam

Guilouz, 2022-11-03:
There is also a new branch for crownest but I have not tested it : https://github.com/mainsail-crew/crowsnest/tree/ref/installer

piwi3910, 2022-11-03:
yes looking at the commits, they are planning on taking the changes i mentioned, and make them configurable during make install, but until that moment you will have to the changes manually for now. 
crowsnest is actually really good. the v42ctl integration is great and allows you to control things like autofocus

Guilouz, 2022-11-03:
Just updated doc with new Crowsnest.

closed 2022-11-03 by Guilouz