Guilouz/Klipper-Flsun-Speeder-Pad #17: timelapse indique DETACHED

gigis57, 2022-11-11:
Bonjour,

J'ai installé par 2x la pad (3 instances) et quand j'arrive à l'installation TIMELAPSE, j'ai sur mainsail -> timelapse indique DETACHED et même le rafraichissement ne fonctionne pas et ni soft ni hard recovery.

Qu'est-il possible de faire ? Merci

[image]

Guilouz, 2022-11-11:
Est-ce que tu as redémarré entre temps ?

gigis57, 2022-11-11:
oui j'ai redemarré entre chaque étape demandé. et comme dis plus je fais et refais l'installation 2x de suite car la 1ière je pensais avoir oublié un truc mais j'ai copié/collé 1 à 1 les étapes décrites. (3 instances)

gigis57, 2022-11-13:
Bien apparemment, je l'ai résolu en supprimant manuellement le dossier timelapse et en le réinstallant. Je poursuis les étapes suivantes. Merci

closed 2022-11-13 by gigis57