Guilouz/Klipper-Flsun-Speeder-Pad #21: Extruder doesn't work

ArikMerzon1, 2022-11-22:
I've followed through with the tutorial and everything seems to be fine.
I did all the calibrations.
Started a new print and although the print head is moving the extruder doesn't work.
I've tried to extrude manually and it's the same thing no extrusion and no errors in the console.
I checked the cables and they are fine.

I can provide more data 

*as a side note, the extruder was working

Guilouz, 2022-11-22:
which printer?

ArikMerzon1, 2022-11-22:
Speed Racer with a Nano V3 board

Guilouz, 2022-11-22:
Do you use the correct printer.cfg file for it?

ArikMerzon1, 2022-11-22:
I've uploaded the `printer.cfg` with the rest of the Configurations files
But now I see that I have multiple printer-####_####.cfg files
Is it correct?

[image]

Guilouz, 2022-11-22:
Stock folder?

Yes it’s normal it’s backup files.

ArikMerzon1, 2022-11-22:
you mean this `/Klipper-Flsun-Speeder-Pad-main/Configurations/FLSUN Super Racer/Robin Nano 3.0-3.1 - LGX Lite`

Guilouz, 2022-11-22:
Do you have LGX Lite extruder ?

ArikMerzon1, 2022-11-22:
sorry, my bad it's stock extruder 
this is my `printer.cfg`
`#   _____             __ _                       _   _                
#  / ____|           / _(_)                     | | (_)                
# | |     ___  _ __ | |_ _  __ _ _   _ _ __ __ _| |_ _  ___  _ __  ___ 
# | |    / _ \| '_ \|  _| |/ _` | | | | '__/ _` | __| |/ _ \| '_ \/ __|
# | |___| (_) | | | | | | | (_| | |_| | | | (_| | |_| | (_) | | | \__ \
#  \_____\___/|_| |_|_| |_|\__, |\__,_|_|  \__,_|\__|_|\___/|_| |_|___/
#                           __/ |                                      
#                          |___/    
#
# Klipper Configurations - Flsun Super Racer      
#
# Guislain Cyril

# MKS Robin Nano V3.0/V3.1
# When running "make menuconfig"
#
# [*] Enable extra low-level configuration options
#     Micro-controller Architecture (STMicroelectronics STM32)  --->
#     Processor model (STM32F407)  --->
#     Bootloader offset (48KiB bootloader (MKS Robin Nano V3))  --->
#     Clock Reference (8 MHz crystal)  --->
#     Communication interface (USB (on PA11/PA12))  --->
# (250000) Baud rate for serial port
# ()  GPIO pins to set at micro-controller startup (NEW)
#
# Note that the "make flash" command does not work with this board.
# Rename the file out/klipper.bin to Robin_nano_v3.bin, copy it an microSD 
# card, insert it in the printer and restart it.

########################################
# Editable Settings
########################################

# Notes: Some settings can be enabled or disabled by removing or adding the '#' symbol
#
# PID (pid_Kp, pid_Ki, pid_Kd) --> [extruder] and [heater_bed] sections
# E-Steps Extruder (rotation_distance) --> [extruder] section --> full_steps_per_rotation * microsteps / steps_per_mm = rotation_distance
# Pressure Advance (pressure_advance) --> [extruder] section -- See: https://www.klipper3d.org/Pressure_Advance.html
# Firmware Retraction --> [firmware_retraction] section -- Requires "Klipper Settings Plugin" for Cura -- See: https://github.com/jjgraphix/KlipperSettingsPlugin
# ADXL345 function for resonance testing --> Enable/Disable [include adxl345_pico.cfg] or [include adxl345_fysetc.cfg] -- Configuration in [input_shaper] section -- See: https://www.klipper3d.org/Measuring_Resonances.html
# NeoPixels function --> Enable/Disable [include neopixels.cfg]
# Timelapse function --> Enable/Disable [include timelapse.cfg]

########################################
# Included Files
########################################

[include macros.cfg]
#[include adxl345_pico.cfg]  #Enable if you want to use ADXL with Rapsberry Pi Pico
#[include adxl345_fysetc.cfg]  #Enable if you want to use ADXL with Fysetc Portable Input Shaper
#[include timelapse.cfg]  #Enable if you want to use Timelapse
#[include neopixels.cfg]  #Enable if you want to use Neopixels

########################################
# Printer Settings
########################################

[printer]
kinematics: delta
max_velocity: 300
max_accel: 4000
max_accel_to_decel: 2000
square_corner_velocity: 5
max_z_velocity: 300
print_radius: 132
minimum_z_position= -5
#delta_radius: 151.7

[... 304 lines cut ...]
#*# height1 = 16.85
#*# height1_pos = 31248.333,31248.333,22588.333
#*# height2 = 16.85
#*# height2_pos = 25087.000,34601.000,25087.000
#*# height3 = 16.85
#*# height3_pos = 22820.333,30416.333,30416.333
#*# height4 = 16.85
#*# height4_pos = 24903.333,24903.333,31711.000
#*# height5 = 16.85
#*# height5_pos = 29525.667,22929.333,29525.667
#*# height6 = 16.85
#*# height6_pos = 32986.667,24910.667,24910.667
#*#
#*# [probe]
#*# z_offset = 15.550
#*#
#*# [endstop_phase stepper_a]
#*# trigger_phase = 45/64
#*#
#*# [endstop_phase stepper_b]
#*# trigger_phase = 55/64
#*#
#*# [endstop_phase stepper_c]
#*# trigger_phase = 60/64
`

ArikMerzon1, 2022-11-22:
and i did use the right folder `/Klipper-Flsun-Speeder-Pad-main/Configurations/FLSUN Super Racer/Robin Nano 3.0-3.1 - Stock`

closed 2022-11-23 by ArikMerzon1

ArikMerzon1, 2022-11-23:
Redid all the steps once again and it's working.
Probably I've missed something along the way.

@Guilouz One question though, is there a way (config) to turn off the small screen of the SR?

Guilouz, 2022-11-23:
Just disconnect it