Guilouz/Klipper-Flsun-Speeder-Pad #24: Error on last step converting the firmware to update the V400

tokamac, 2022-12-01:
After the following step which completes successfully (after all previous successful steps):
```
pi@speeder-pad:~/klipper$ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/spi_software.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.11.0-0-ge02b7256
  Linking out/klipper.elf
  Creating hex file out/klipper.elf.hex
```
Th step converting the firmware fails:
```
pi@speeder-pad:~/klipper$ ./scripts/update_mks_robin.py out/klipper.bin out/Robin_nano35.bin
Traceback (most recent call last):
  File "./scripts/update_mks_robin.py", line 38, in <module>
    main()
  File "./scripts/update_mks_robin.py", line 25, in main
    f = open(infilename, "rb")
IOError: [Errno 2] No such file or directory: 'out/klipper.bin'
```

Guilouz, 2022-12-01:
You have not selected correct firmware config.
You obtain a hex file and not a bin file.

closed 2022-12-01 by Guilouz

tokamac, 2022-12-02:
I think I see where I made my mistake: I've just selected "Enable extra low-level configuration options" without touching the other subsettings like micro-controller, etc.
Will check and come back to close the issue.

tokamac, 2022-12-02:
That was it! Incorrect options selected at the step `make menuconfig`
Problem solved. 
I close the (non)issue.