Guilouz/Klipper-Flsun-Speeder-Pad #32: Problème connexion wifi

Flazor-sun, 2022-12-23:
Bonjour, 

Je me permets de vous contacter car j'ai fait la mise à niveau de mon imprimante V400. Tout c'est bien passé.
Sauf que j'ai voulu la changer de wifi (après m-à-j) est la, impossible de la connecter au réseau voulu, j'ai le message Attempting to connect to (SSID).
J'ai donc supprimer le réseau sur lequel je voulais la connecter en laissant l'ancien, puis redémarré. Elle ne se reconnecte même pas à l'ancien. 
Alors j'ai essayé de supprimer l'ancien et reconnecté, toujours pareil j'ai le même défaut qu'expliqué ci-dessus (attempting).
Cela serait-il dû modification de l'emplacement wifi?

Merci d'avance

[image]

Guilouz, 2022-12-23:
Problème non lié au github. Pour la prochaine fois merci de ne reporter que les problème lié au contenu du github.

La partie wifi de l'OS du pad est vraiment merdique. Quelqu'un qui a changé de box a eu le même tour, il a dû réinstaller le pad.

closed 2022-12-23 by Guilouz

Flazor-sun, 2022-12-23:
Très bien c'est noté, merci de la rapidité.
Bonnes fêtes ;)