Guilouz/Klipper-Flsun-Speeder-Pad #44: USB ADXL345 Triangle Lab

JSI33, 2023-01-17:
Hello. I have seen that you make an STL to put an USB ADXL345 TriangleLab on a V400. But on your github you don't explain how to configure this adxl and i trying to flash it with the TriangleLab instructions but it fail when i try to flash it. Can you help me please?

Thanks

JSI33

Guilouz, 2023-01-17:
It's explain...

https://github.com/Guilouz/Klipper-Flsun-Speeder-Pad#use-adxl345-with-fysetc-portable-input-shaper

closed 2023-01-17 by Guilouz

JSI33, 2023-01-18:
Thanks for your answer. Yes bute the TriangleLab adxl345 have two buttons. The manipulation seems to be different.
[image]

Guilouz, 2023-01-18:
Ah sorry i didn’t see it’s a triangle lab.

Check trianglelab documentation for that: https://drive.google.com/drive/mobile/folders/1lXzN1LilcKusdOj9-mwBFJ6B84XiJpRR?spm=a2g0o.detail.1000023.102.74906a1fAo5Wk1&pli=1

JSI33, 2023-01-18:
Done! But i will be missing somthing cause it didn't work.

Guilouz, 2023-01-18:
I didn’t have it so i can’t help you, sorry.