Guilouz/Klipper-Flsun-Speeder-Pad #56: Unable to recovery klipper

itajackass, 2023-01-23:
[image]

Today, Every time i poweron speeder pad i get INVALID klipper message. I tried hard and soft recovery without success... It work about 10 minutes on this screen (see pic), then fail.

itajackass, 2023-01-24:
after some retry i get klipper valid again. before this problem I've assigned fixed IP of speederpad into my network to 192.168.1.102........now when i power on speederpad I see into my network 2 different devices with same name SPEEDERPAD and 2 different MAC ADDRESS and of course 2 different IP.....192.168.1.102 (fixed ip...but not reachable)....and 192.168.1.106 (dhcp....working)......Why?

Guilouz, 2023-01-25:
No support here. Please only post issues related to github.

closed 2023-01-25 by Guilouz